ivanseidel/LinkedList #24: Add unit tests and continuous integration

scls19fr, 2018-05-02:
Hello,

I'm considering this library as a dependency  for an application.
Althought I'd prefer to rely on a library which have unit tests.

You can find some links dealing with continuous integration and Arduino
https://github.com/adafruit/travis-ci-arduino
https://www.pololu.com/blog/654/continuous-testing-for-arduino-libraries-using-platformio-and-travis-ci
(they are using http://platformio.org/ )

Kind regards

PS : see also https://docs.travis-ci.com/user/integration/platformio/

https://github.com/platformio/platformio-remote-unit-testing-example

scls19fr, 2018-05-02:
See also https://github.com/mmurdoch/arduinounit/

Just for unit tests (not CI although) https://github.com/mmurdoch/arduinounit/issues/63

ivanseidel, 2019-04-16:
This got merged!

closed 2019-04-16 by ivanseidel