ivanseidel/LinkedList #28: non-virtual deconstructor

RyanAveryBSU, 2018-10-15:
-wall using this class gives me a message about calling delete on a linkedlist<obj*>* about a non-virtual deconstructor which leads to a undetermined state

RyanAveryBSU, 2018-10-15:
edit: this is the warning I am getting when I call delete(myClassList);

warning: deleting object of polymorphic class type 'LinkedList<MyClass*>' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor] delete(myClassList);

After some googleing it seems that its not "might cause" but actually "100% does cause" undefined behaviour

Raincode, 2019-09-04:
Does MyClass have a virtual destructor? I think this issue can be closed, it is almost a year old and we haven't heard anything new since.

ruigaspar, 2021-03-03:
I added a pull request to fix this: https://github.com/ivanseidel/LinkedList/pull/45

closed 2022-03-10 by ivanseidel

Raincode, 2022-05-14:
> I added a pull request to fix this: #45

I may be wrong (I haven't been programming for a few years...), but I'm pretty sure the above issue was wrongful use of the LinkedList as opposed to a problem with LinkedList itself. Of course, we'll never be 100% sure, since no example code was provided.
But IMHO (with my limited knowledge), I don't think the LinkedList destructor needs to be virtual here.
And allocating LinkedList itself on the heap seems pointless to me... Since the whole point of LinkedList is to manage memory internally and allocate its nodes on the heap itself.