ivanseidel/LinkedList #32: sort for class objects?

michelep, 2019-08-27:
How use sort for class objects?

Raincode, 2019-09-04:
the sort member function takes the comparison function for two Objects of type T as an argument:
`int (*cmp)(T &, T &)`

simply define a function which does this for your class. I'm not sure, but I'm guessing the return value works similar to strcmp.

```
class Foo {
// ...
};

int compare_Foo(const Foo& lhs, const Foo& rhs) {
  // ...
}

LinkedList<Foo> list;
// ...
list.sort(compare_Foo);
```

P.S.: The signature should be changed to take const T&, since comparing musn't modify objects.

I don't know if Arduino supports operator overloading, but overloading ==, <, > etc. would be preferable imho. However this would require changes in the implementation of sort as well.

ivanseidel, 2022-03-10:
I suppose arduino doesnt support sorting like that, but would be nice to see it working on other platforms!

closed 2022-03-10 by ivanseidel