ivanseidel/LinkedList #33: Compile errors with 1.8.10 arduino ide

mdlagor, 2019-10-09:
Hello,
I get these compilation errors using your lib in Arduino 1.8.10 ide:

home/user/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Transition*]':
/home/user/Arduino/libraries/LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = Transition*]'
/home/user/Arduino/libraries/StateMachine/src/State.h:83:32:   required from here
/home/user/Arduino/libraries/LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<Transition*>*' in return
  return false;
         ^~~~~
/home/user/Arduino/libraries/LinkedList/LinkedList.h: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = State*]':
/home/user/Arduino/libraries/LinkedList/LinkedList.h:314:28:   required from 'T LinkedList<T>::get(int) [with T = State*]'
/home/user/Arduino/libraries/StateMachine/src/StateMachine.h:54:41:   required from here
/home/user/Arduino/libraries/LinkedList/LinkedList.h:160:9: error: cannot convert 'bool' to 'ListNode<State*>*' in return

With Arduino ide 1.8.9 these errors are warnings and I can compile but with the new version 1.8.10 I can't
Thank you very much and best regards

burner-, 2019-10-28:
Looks that arduino ide provide old broken version of lib instead of current git version.

spymastermatt, 2020-02-12:
@burner- is correct, the latest version on git fixes this issue. The arduino IDE ships with the latest _release_ (1.2.3 as i write this), which does not contain the fix.
An easy fix while you wait is to find line 160 in LinkedList.h (/home/user/Arduino/libraries/LinkedList/LinkedList.h for the OP) and change _return false_ to _return NULL_

strykeroz, 2020-07-04:
IDE 1.8.13 still has this issue.  Thanks for resolving it here.

UUID-2414, 2021-04-10:
Had the same errors as mdlagor, an update from GitHub fixed it. I guess the IDE still has an older version as of now, 10.04.2021
Thanks for pointing us towards the GitHub here.

ivanseidel, 2022-03-10:
Should be fixed

closed 2022-03-10 by ivanseidel