ivanseidel/LinkedList #36: Compile error for Arduino Nano

SummerTime244, 2020-02-10:
Hello,
I'm trying to use your library on Arduino IDE 1.8.10 with an Arduino Nano Board but I get the following error message :

`Arduino: 1.8.10 (Windows 7), Board: "Arduino Nano, ATmega328P (Old Bootloader)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\EESM\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\EESM\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\EESM\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\EESM\AppData\Local\Temp\arduino_build_776334 -warnings=none -build-cache C:\Users\EESM\AppData\Local\Temp\arduino_cache_884885 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\EESM\Documents\Arduino\test_arg_list_linked\test_arg_list_linked.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\EESM\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\EESM\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\EESM\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\EESM\AppData\Local\Temp\arduino_build_776334 -warnings=none -build-cache C:\Users\EESM\AppData\Local\Temp\arduino_cache_884885 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\EESM\Documents\Arduino\test_arg_list_linked\test_arg_list_linked.ino
Using board 'nano' from platform in folder: C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2
Using core 'arduino' from platform in folder: C:\Users\EESM\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2
Detecting libraries used...
"C:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\eightanaloginputs" "C:\\Users\\EESM\\AppData\\Local\\Temp\\arduino_build_776334\\sketch\\test_arg_list_linked.ino.cpp" -o nul
Alternatives for LinkedList.h: [LinkedList@1.3.1]
ResolveLibrary(LinkedList.h)
  -> candidates: [LinkedList@1.3.1]
"C:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\eightanaloginputs" "-IC:\\Users\\EESM\\Documents\\Arduino\\libraries\\LinkedList" "C:\\Users\\EESM\\AppData\\Local\\Temp\\arduino_build_776334\\sketch\\test_arg_list_linked.ino.cpp" -o nul
"C:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\EESM\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\eightanaloginputs" "-IC:\\Users\\EESM\\Documents\\Arduino\\libraries\\LinkedList" "C:\\Users\\EESM\\Documents\\Arduino\\libraries\\LinkedList\\tests.cpp" -o nul
Alternatives for iostream: []
ResolveLibrary(iostream)
  -> candidates: []
C:\Users\EESM\Documents\Arduino\libraries\LinkedList\tests.cpp:5:10: fatal error: iostream: No such file or directory

 #include <iostream>

          ^~~~~~~~~~

compilation terminated.

Multiple libraries were found for "LinkedList.h"
 Used: C:\Users\EESM\Documents\Arduino\libraries\LinkedList
Using library LinkedList at version 1.3.1 in folder: C:\Users\EESM\Documents\Arduino\libraries\LinkedList 
exit status 1
Error compiling for board Arduino Nano.`

CodeMedve, 2020-02-11:
Try an older version, for me worked.
[https://www.arduinolibraries.info/libraries/linked-list](https://www.arduinolibraries.info/libraries/linked-list)

SummerTime244, 2020-02-12:
Thank you ! It worked with Arduino 1.0.6.

closed 2020-02-12 by SummerTime244