ivanseidel/LinkedList #39: iostream compilation error on Arduino Mega

Luana-Menezes, 2020-06-06:
/home/luana/Arduino/libraries/LinkedList/tests.cpp:5:10: fatal error: iostream: No such file or directory
 #include <iostream>
          ^~~~~~~~~~
compilation terminated.
exit status 1
Erro compilando para a placa Arduino Mega or Mega 2560

morrocoyo08, 2020-08-27:
Idem com Uno IDE 1.8.13

mccartes, 2021-07-01:
It looks like Iostream is not supported in arduino: 

This fails to compile on arduino because the linked list library includes the following in the "tests.cpp" file:
#include <iostream>
std::cout<< "Tests pass"<< std::endl;

when I comment out these two lines, then the whole thing compiles successfully

GregJohnStewart, 2021-08-24:
> Deleted _test.cpp_, it fixed the issue. Problem solved.

Confirmed, worked for me as well. Fairly annoying for someone trying to use the lib with Arduino, but at least it's not a complex fix. Looks to me that the Arduino IDE is being overzealous and just compiling everything in a library's source directory. I wonder if there is a way to get around that?

closed 2022-03-10 by ivanseidel