ivanseidel/LinkedList #52: Including LinkedList.h breaks Nano 33 IoT setup method

Tetrahedran, 2021-09-21:
I wrote a simple sketch that essentially lets the buildin LED blink through the loop method. At the moment I include LinkedList.h evering breaks and the Arduino does not exit setup-method (visible via the not flashing LED).
My test code:

#include<LinkedList.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);  
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
}

per1234, 2022-03-10:
This may have been fixed by https://github.com/ivanseidel/LinkedList/pull/48

Tetrahedran, 2022-03-10:
Sadly not. I downloaded version 1.3.3 from Github, installed it according to the instructions in my lib folder, restarted the Arduino IDE and uploaded the same sketch but including LinkedList.h still breaks everything.