ivanseidel/LinkedList #53: Install Instructions Are Incorrect

Warren52, 2021-09-21:
The current install instructions need to have an additional step included
when used with Arduino (at least 1.8.13). Without this step, the test.cpp
file will be included when compiling a script that includes LinkedList. 
The result will be a second "main()" function, from test.cpp, included in
the compile which will override the correct "main()" function for the chosen 
board / MCU. This means the normal "setup()" and "loop()" functions will NOT
run making it appear like LinkedList has trashed the program or the program
cannot work. 

The solution is quite simple. After the third step of pasting the modified
folder to the Library folder, add:

  3a - In the LinkedList Library folder, rename test.cpp to test.cpp_hidden
  
So the resulting steps are:

1. Download the Latest release from gitHub.
2. Unzip and modify the Folder name to "LinkedList" (Remove the '-version')
3. Paste the modified folder on your Library folder (On your Libraries folder inside Sketchbooks or Arduino software).
**3a. In the LinkedList Library folder, rename test.cpp to test.cpp_hidden**
4. Reopen the Arduino software.

jonathonball, 2021-12-07:
I also encountered this issue.  My solution was to delete test.cpp.

emergencyhamnet, 2022-02-27:
Good help, I had same issue would not compile.  missing correct file.
I deleted the file test.cpp from the LinkedList library and all seems well now.

ivanseidel, 2022-03-10:
This should be fixed in 1.3.3, can you verify?

Warren52, 2022-04-04:
Sorry for the delay - was involved with another project. 
Installed 1.3.3 and created a LinkedList.ino file from your Test.cpp. Built it and programmed a Teensy 3.2. All tests passed and did not have to change any file names or anything. Nice job - thanks!

FYI - If you want, my LinkedList.ino code follows:

// --------------------------------------------------------------------------
//
//  Title:
//    LinkedList.ino - LinkedList Test.cpp turned into an .ino file
//
// Purpose:
//    I had entered a LinkedList github issue: "Install Instructions Are
//    Incorrect". The maintainer got back to me with an updated version
//    of LinkedList (1.3.3). I thought I would make sure everything worked
//    and copied his Test.cpp it to this file, renamed it, modified some
//    conntent, programmed a Teensy 3.2, then ran it. All worked well.
//
//
//  History:
//    ddmmmyyyy   Author    Description
//    ---------   -------   -----------
//    04apr2022   W. Holt
//      Initial creation from LinkedList Test.cpp
//      When this was compiled and run on a Teensy 3.2, all tests passed
//
//
// --------------------------------------------------------------------------

// ------- includes
#include <LinkedList.h>
// #include <assert.h>    -- cannot use in an embedded world
// #include <iostream>    -- using Serial.print...

void GivenNothingInList_WhenSizeCalled_Returns0()
{
    //Arrange
    LinkedList<int> list = LinkedList<int>();

    //Act Assert
    Serial.print("list.size() == 0): ");
    Serial.println(list.size() == 0);
}

void GivenNothingInList_WhenAddNodeAtPlace0_ReturnsTrue()
{
    //Arrange
    LinkedList<int> list = LinkedList<int>();

    //Act Assert
    Serial.print("list.add(0, 1) == true): ");
    Serial.println(list.add(0, 1) == true);
}

void GivenOneNodeInList_WhenAddNodeIndexLargerThanListSize_ThenNodeAddedToEndOfList()
{
    //Arrange
    LinkedList<int> list = LinkedList<int>();
    list.add(1);

    //Act
    list.add(5, 2);
    list.add(4, 3);

    //Assert
    Serial.print("list.get(0) == 1): ");
    Serial.println(list.get(0) == 1);
    Serial.print("list.get(1) == 2): ");
    Serial.println(list.get(1) == 2);
    Serial.print("list.get(2) == 3): ");
    Serial.println(list.get(2) == 3);
[... 332 lines cut ...]
    case 11: GivenTwoNodesInList_WhenPopIsCalled_ThenListIsShorter(); break;
    case 12: GivenNothingInList_WhenShiftIsCalled_ThenReturnsFalse(); break;
    case 13: GivenOneNodeInList_WhenShiftIsCalled_ThenReturnsData(); break;
    case 14: GivenOneNodeInList_WhenShiftIsCalled_ThenListEmpty(); break;
    case 15: GivenThreeNodesInList_WhenShiftIsCalled_ThenReturnsFirstData(); break;
    case 16: GivenThreeNodesInList_WhenShiftIsCalled_ThenListIsShorter(); break;
    case 17: GivenNothingInList_WhenRemoveIsCalled_ThenFalseIsReturned(); break;
    case 18: GivenThreeNodesInList_WhenRemoveIsCalledAtPlace0_ThenFirstNodeDataIsReturned(); break;
    case 19: GivenThreeNodesInList_WhenRemoveIsCalledAtPlace2_ThenLastNodeDataIsReturned(); break;
    case 20: GivenThreeNodesInList_WhenRemoveIsCalled_ThenListIsShorter(); break;
    case 21: GivenNothingInList_WhenGetIsCalled_ThenReturnsFalse(); break;
    case 22: GivenThreeNodesInList_WhenGetIsCalled_ThenReturnsData(); break;
    case 23: GivenNothingInList_WhenClearIsCalled_ThenSizeUnchanged(); break;
    case 24: GivenThreeInList_WhenClearIsCalled_ThenListEmpty(); break;
    default:
      if(ndx > 25)
        ndx = 25;
      break;
    }

  // ------- delay, led off, update ndx
  delay(200);
  digitalWrite(LED_BUILTIN,LOW);
  ndx++;
}

closed 2022-04-04 by Warren52