ivanseidel/LinkedList #57: Arduino auto-reboot 😄

maincraft-io, 2021-12-10:
Simple code below doing auto-reboot of Arduino:
```
#include <Arduino.h>
#include <DEVICE.h>
#include <elapsedMillis.h>
elapsedMillis testTimer;

void setup()
{
  
  Serial.begin(115200);
  Serial.println("Working again");

}

void loop()
{

  if(testTimer > 50) 
  {
    LinkedList<byte> * packet = new LinkedList<byte>();
    packet->add(0xF1);
    packet->add(0x00);
    packet->add(0x01);
    packet->add(0x00);
    packet->add(0x0D);
    packet->clear();
    testTimer = 0;
  }
}
```
```
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Working again
Working again
Working again
Working again
Working again
Working again
Working again
Working again
```

vortigont, 2021-12-10:
Your example is leaking memory,
set it like this and you'll be fine

```
LinkedList<byte> * packet;
void setup()
{
    packet = new LinkedList<byte>();
}

void loop()
{
  if(testTimer > 50) {
     packet->add(0xF1);
     // etc...
   }
}

```

maincraft-io, 2021-12-10:
> Your example is leaking memory, set it like this and you'll be fine
> 
> ```
> LinkedList<byte> * packet;
> void setup()
> {
>     packet = new LinkedList<byte>();
> }
> 
> void loop()
> {
>   if(testTimer > 50) {
>      packet->add(0xF1);
>      // etc...
>    }
> }
> ```

Amazing! Thanks for fastest reply and help. 
Very appricate for idea.

closed 2021-12-10 by maincraft-io