ivanseidel/LinkedList #58: Example won't compile for Arduino Portenta

njhetherington, 2021-12-21:
Compile command: `arduino-cli compile ~/Arduino/libraries/LinkedList/examples/ClassList/ClassList.pde -b arduino:mbed_portenta:envie_m7 --libraries ~/Arduino/libraries/`
Errors:
"
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino: In function 'void setup()':
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:49:15: error: 'class LinkedList<Animal*>' has no member named 'add'
  myAnimalList.add(cat);
               ^~~
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:50:15: error: 'class LinkedList<Animal*>' has no member named 'add'
  myAnimalList.add(emu);
               ^~~
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:51:15: error: 'class LinkedList<Animal*>' has no member named 'add'
  myAnimalList.add(dog);
               ^~~
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino: In function 'void loop()':
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:57:28: error: 'class LinkedList<Animal*>' has no member named 'size'
  Serial.print(myAnimalList.size());
                            ^~~~
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:62:34: error: 'class LinkedList<Animal*>' has no member named 'size'
  for(int i = 0; i < myAnimalList.size(); i++){
                                  ^~~~
/home/cellula/Arduino/libraries/LinkedList/examples/ClassList/ClassList.ino:65:25: error: 'class LinkedList<Animal*>' has no member named 'get'
   animal = myAnimalList.get(i);
                         ^~~

Error during build: exit status 1
"

Arduino CLI version 0.20.2
Arduino Portenta core version 2.6.1
LinkedList library version 1.3.2

Compiles for Arduino Mega after deleting _tests.cpp_.

ivanseidel, 2022-03-10:
can you paste the code?

njhetherington, 2022-03-10:
I was trying to compile the ClassList example in your repo.
> Compile command: `arduino-cli compile ~/Arduino/libraries/LinkedList/examples/ClassList/ClassList.pde -b arduino:mbed_portenta:envie_m7 --libraries ~/Arduino/libraries/`

julianstj1, 2024-06-17:
This is happening because the mbed libraries have a LinkedList structure as well, with less capabilities than this library. 

I fixed it by manually editing the LinkedList.h file in this library to LinkedList_git.h and all the references inside the file to LinkedList_git instead. I then put this header file in the directory of my .ino and it works.

callmep0l, 2024-06-18:
Hello @julianstj1 thanks a lot for sharing it! That was correct, this helped me. 

Can I ask how you found/knew out this? Also, what was your methodology for discovering that mbed libraries have a LinkedList structure? I'm learning more about micro-controllers and I'm curious to know  because it might help me to have a better understand of how to find more features of micro-controllers :)

Pol.

vortigont, 2024-06-19:
A better way to workaround the issue is in PR #51

julianstj1, 2024-06-19:
I found it just using view definition in the Arduino IDE. I right clicked on my object declaration of LinkedList, and it took me to a LinkedList.h file that was somewhere in the Arduino directories for mbed giga, so I knew it was linking against the wrong file. So just changing the name so it doesn't conflict makes it work.